j4cbo/tiny5-m2-riser #1: Group buy interest?

j4cbo, 2024-04-05:
A few people have mentioned interest in buying one of these so I'm starting a thread to discuss. Very-low-quantity manufacturing isn't cheap (particularly for a controlled-impedance design with specialized connectors like this) so I'd expect something in the $50-$100 USD ballpark range depending on quantity.

galvesribeiro, 2024-04-05:
Hello! I do want one. Sign me in! :)

kitor, 2024-04-06:
I'd build one myself so I'm interested in either bare pcbs or project files.

struthio, 2024-04-07:
hi, I am interested in final product or bare pcbs. I have also 2 questions:
1. When final price will be known ?
2. What is price for shipping ? I saw You are located in USA, so shipping to Europe might be pricey (compared to price of adapter).

marcojk, 2024-04-08:
> hi, I am interested in final product / bare pcbs / project files. I have also 2 questions:
> 
> 1. When final price will be known ?
> 2. What is price for shipping ? I saw You are located in USA, so shipping to Europe might be pricey (compared to price of adapter).

Same here, possibly an Europe-based group buy could be set up.
Opening the design would be nice too!

systemerror-codenotfound, 2024-04-08:
Hi,
It would be possible to try an develop a version for the M910X ?

I'm part of a group here on spain that would be really interested im several parts

skruje, 2024-04-09:
I'd probably be in for 1-3 of these once you're satisfied with the prototypes, depending on the pricing.

Will the custom riser still require all the SMD soldering on the motherboard shown in badger707/m920q-dual-NVME?

Nice to haves for future revs:
- option for an x8 port for shorter cards like the AOC-STGN-I2S. Maybe a jumper to select x8x4 or x4x4x4. If not practical electrically, an x8 slot wired for x4, or just one with an ["open" end](https://upload.wikimedia.org/wikipedia/commons/a/a3/PCIe_J1900_SoC_ITX_Mainboard_IMG_1820.JPG) would be more flexible at similar cost.
- 12V (~0.1A) fan header for cooling hotter cards without additional soldering

kitor, 2024-04-09:
> Hi, It would be possible to try an develop a version for the M910X ?

@systemerror-codenotfound Why don't you start own issue instead of hijacking non-related one?

> Will the custom riser still require all the SMD soldering on the motherboard shown in [badger707/m920q-dual-NVME](https://github.com/badger707/m920q-dual-NVME/)?

@skruje No, you are confusing two different mods. This one requires bifurcation mod which is two resistors change from [badger707/m920q-pcie-bifurcation/issues/1]

If you use both, you can go crazy and have up to 6 NVME drives in this computer. One in existing slot, one in dual nvme mod, three using riser we discuss here and one extra in WiFi slot (but that will work as PCIe x1) :)

> option for an x8 port for shorter cards like the AOC-STGN-I2S. Maybe a jumper to select x8x4 or x4x4x4. 

This was already answered in badger707 thread. It is impossible, not impractical, due to a few factors:

* you can't "switch" where 4 pairs of signals go with a single jumper
  * switching between bifurcated and non-bifurcated slot is done via modding the motherboard - you change cpu configuration resistors for that.
  * in modified state, PCIe lanes are out of order and this riser corrects for that
  * thus in unmodified (x8) state lanes would be out of order again and nothing would work
* you could try wire up signals to both one of NVME m.2 slots and PCIe slot, but this fails quickly:
  * out of order problem from above
  * even if not, signal reflections would be the issue (considering lanes connected to both slots)
  * and in that hypothetical / impossible situation you would still need to select x8+m.2 or x4 + dual m.2 by soldering proper slots into your riser and selecting resistors on the board.

Open end slot seems a good idea on the first glance (but you can also mod your slot very easily with craft knife). However from the photos I'm not sure if x8 card would even fit - I'm under impression that first m.2 slot would interfere anyway.

j4cbo, 2024-04-09:
The way I think of it, https://github.com/badger707/m920q-dual-NVME turns an m920q into roughly an m920x, which has two onboard M.2 connectors for NVMe SSDs. Then my board in conjunction with the two resistors changed in https://github.com/badger707/m920q-pcie-bifurcation adds two more M.2s (one off the PCH, one off the bifurcated CPU port) and provides a PCIe x4 slot. As @kitor says, if you aren’t using the slot for a network card, then (with a PCIe x4 to M.2 board) you could add another SSD there, and if you aren't using the WiFi M.2 you could (with an A+E-key to M-key adapter) add yet another SSD there, albeit with only one PCIe lane.

There's about 10mm vertically between the M.2 connectors and the x4 slot on my board, so a PCIe card that extends under the M.2 SSDs would be fine as long as it didn't have a large heatsink.

j4cbo, 2024-04-10:
Update: price for the first batch will be $70 USD each plus shipping, and I expect them to be available in mid-May. There will also be a 12V fan header on the board with the standard 3-pin PC fan pinout. This is for m920q/m920x only. The m720q doesn't support bifurcation, and the m910 series has a different pinout and lacks the additional 4 PCH lanes on the socket. (A single-M.2 m720q variant is in the works, though.)

Shipping will be about $5 within the US. Internationally, slow shipping (1-4 weeks) is about $25, or by UPS about $40. It might make sense for me to ship a bunch to someone in Europe for further distribution from there.

skruje, 2024-04-10:
Love that @j4cbo managed to get a fan header in there. I'm out for now; hoping we can find a more economical way to deliver to Canada for future batches. Any chance the design will be open sourced?

j4cbo, 2024-04-10:
Canada shipping is cheaper than Europe - I edited the post above with those prices and quantity discounts. I'm not opposed to open-sourcing the design but, selfishly, I'd like to make back what I spent on the first batch first :)

chocmake, 2024-04-11:
> This is for m920q/m920x only. The m720q doesn't support bifurcation

Just to clarify, badger's bifurcation repo states compatibility for M720q. Is something different for this riser or am I missing something?

j4cbo, 2024-04-11:
> Just to clarify, badger's bifurcation repo states compatibility for M720q. Is something different for this riser or am I missing something?

It's mentioned in the README, but there's a good argument as to why bifurcation wouldn't work on the m720q here: https://github.com/badger707/m920q-pcie-bifurcation/issues/2#issuecomment-1835940720

The m720q does still have the extra 4 lanes from the PCH, though, which should support a single M.2.

j4cbo, 2024-04-18:
I have one prototype available which I'm offering for free (within the US) if anyone wants to beta test. It lacks the fan header and open-back PCIe slot that the production ones will have (probably a month out), and it's not pretty, but I've tested in my own server and it is working well. My email is on my Github profile.

galvesribeiro, 2024-04-18:
Would test for you but I'm in Brazil :P

j4cbo, 2024-04-19:
Update: I've placed the production order :) I was able to get the price down to $50. The production boards will include an open-backed PCIe slot and a 12V fan header, and will be ready in about a month.

galvesribeiro, 2024-04-19:
Looking forward for it :)

struthio, 2024-04-19:
@j4cbo $50 for M920Q one ? (2x NVME + 1xPCIe for Ethernet) ? If Yes then I even consider getting 2.
Are You taking any pre-orders ?

galvesribeiro, 2024-04-19:
Yeah, rephrasing, I want 2 as well 😄 

@j4cbo if you can send email and/or PayPal/Wise or whatever way you decide to be paid, we can put the pre-order ahead.

j4cbo, 2024-04-21:
To pre-order, send me an email - my email address is shown on my main Github profile :)

galvesribeiro, 2024-04-21:
Done :shipit:

struthio, 2024-04-22:
> I'm not opposed to open-sourcing the design but, selfishly, I'd like to make back what I spent on the first batch first :)

Maybe then, add project files as a documentation to bought boards or something like that ?

j4cbo, 2024-04-29:
Just an update here, the manufacturer has finished fabricating the bare PCBs and is now preparing for assembly. Still looking like I'll have them by the end of May.

mdronald, 2024-05-01:
> Just an update here, the manufacturer has finished fabricating the bare PCBs and is now preparing for assembly. Still looking like I'll have them by the end of May.

I'm interested in buying one or two but I've the m920q model. Would I have to mod it using [badger707](https://github.com/badger707/m920q-dual-NVME/commits?author=badger707) guide to work as a m920x before using your riser?

j4cbo, 2024-05-01:
> I'm interested in buying one or two but I've the m920q model. Would I have to mod it using [badger707](https://github.com/badger707/m920q-dual-NVME/commits?author=badger707) guide to work as a m920x before using your riser?

This is independent of the m920q-to-m920x conversion; it'll still give you two extra M.2 slots. It does require installing the two resistors described in https://github.com/badger707/m920q-pcie-bifurcation/issues/1 though.

m90q, 2024-05-04:
Hi, I have a M90q Gen3 ; will this mod work ? if so How can I order one?

kitor, 2024-05-04:
No, this will not work.

j4cbo, 2024-05-04:
> Hi, I have a M90q Gen3 ; will this mod work ? if so How can I order one?

Unfortunately not - the M90q riser has a different pinout.

m90q, 2024-05-04:
> > Hi, I have a M90q Gen3 ; will this mod work ? if so How can I order one?
> 
> Unfortunately not - the M90q riser has a different pinout.

thanks.

m90q, 2024-05-04:
> No, this will not work.

thanks

j4cbo, 2024-05-07:
Update: assembly is done ahead of schedule, and they're being prepared to ship to me now :)

badger707, 2024-05-10:
Super cool project! I'll drop you a quick email.

galvesribeiro, 2024-05-15:
Just got mines! Looking forward to get back home and try them out!

[image]

crimeofheart, 2024-05-15:
This is an excellent project! @j4cbo did you get any orders from Europe? If so I am ready to buy one and maybe two in case someone else wants to buy with less shipping cost (so that I can send it after receiving). I don't need them per my current setup, however I want to show my support for such great projects within my budget. I will send you an email.

j4cbo, 2024-05-15:
@struthio is also in Europe and had mentioned interest in a group order. The shipping cost is close to the same regardless of the number of boards, so with 5 boards it becomes a lot more cost effective :)

marcojk, 2024-05-16:
I'm in for 2 boards too.
Would it be possibile for j4cbo to create an new batch (how many boards in each one?) and send it firsthand to someone in Europe? And then from here everyone will get his. 90-100€ is a bit pricey for my budget.

crimeofheart, 2024-05-16:
Hey @j4cbo, my emails to you are calssified as spam. Tried to send a couple of emails, not sure whether you will eventually receive them :)
I am willing to support and can receive the bulk order, then distribute. I am located in Spain. Shipping from here to any EU country varies between 8 to 20 EUR depending on urgency, tracking etc. However on the other hand I don't want to endup with batch of risers that people didnt commit in the end. Not sure how this can be solved.

@badger707 I believe the import tax can be avoided if the box is declared as a gift/prototype. 
This way the cost of the riser should be around (50 usd x number_of_item + 20 usd  shipping from my side). I calculated the 20 USD shipping as:  30 USD from US / number of people in Europe (currently 4) + shipping from my side. All these are theorotical and can vary +- 5 usd per card I beleieve.

struthio, 2024-05-17:
@j4cbo Thanks for mentioning me :)

@crimeofheart "I believe the import tax can be avoided if the box is declared as a gift/prototype." - this is true if You get 1-2 items. If this will be like bulk order, like 10-20 exact items in onebox, there is high probability that customs will just calculate tax (no matter what is marked in shipping declaration). 

And regarding shipping, can You check exact prices (I mean +/- 5USD) ? If this is +5 USD then total shipping is 25USD which is same as 'cheaper' option mentioned by @j4cbo directly from USD, so I do not think it is worth a hassle to repack it in EU.

Please also consider need to repack (so You providing additional boxes, fillers for packages etc).

If anyone is from PL, I can redistribute it within PL shipping in PL (InPost) is in range of 3-4USD.

kitor, 2024-05-17:
> If anyone is from PL, I can redistribute it within PL shipping in PL (InPost) is in range of 3-4USD.

Yup, I am. Still waiting to see what the _total estimated damages_ is. 

I wonder where @j4cbo ordered the first batch. Maybe it would be just cheaper to order a second batch directly to someone in EU ... assuming there's enough interest of course.

marcojk, 2024-05-17:
> I wonder where @j4cbo ordered the first batch. Maybe it would be just cheaper to order a second batch directly to someone in EU ... assuming there's enough interest of course.

This is the option i was referring to. Collect the order, pay to j4cbo, order shipped directly to europe and from here we will manage somehow :)

j4cbo, 2024-05-17:
The PCB fabrication and soldering was done by a contract manufacturer in China, but then there's a programmable logic device that I need to flash afterwards. (I'm very glad I designed it that way - it allowed me to fix a compatibility issue with x8/x16 cards in the open-backed slot that I only discovered after the boards had been sent for fabrication!)

I think the best bet might just be for me to ship individually with USPS First Class, which is potentially slow (they say 1-4 weeks) but would only be $17.50 (about 16€) per shipment. Plus that way each shipment stays relatively low-value in terms of duty thresholds. With two boards in a shipment that's around 110€ total for two boards, plus taxes... not too bad?

wociscz, 2024-05-17:
EU (CZ) here. Count me in for 2 pcs (for the next batch if this is already sold out).

j4cbo, 2024-05-17:
@badger707 @kitor and @wociscz , could you each send me an email? Address is in my Github profile :)

badger707, 2024-05-17:
@j4cbo it is the one I've sent you on Saturday 11th, saw your email response bud didn't managed to reply yet.

struthio, 2024-05-23:
> > If anyone is from PL, I can redistribute it within PL shipping in PL (InPost) is in range of 3-4USD.
> 
> Yup, I am. Still waiting to see what the _total estimated damages_ is.
> 
> I wonder where @j4cbo ordered the first batch. Maybe it would be just cheaper to order a second batch directly to someone in EU ... assuming there's enough interest of course.
> 
> BTW: for domestic shipping you can also check DPD. For Pickup - Pickup shipments with 40SS24 code you can go right now as low as ~7,50PLN for small parcels :)

Hi,

I've sent You an email, Check if it didn't went to "Spam" folder :)

wociscz, 2024-05-31:
Raisers arrived. Total cost per piece is ~$83 with vat and customs (Czech Republic). Looks great/professional grade product. Thanks!

j4cbo, 2024-06-04:
Still have 6 available :)

RoxOnix, 2024-06-08:
Hi,
I'll take 2, check your email.

su-thomas, 2024-06-12:
Hi, interested in buying, email sent

neutronstriker, 2024-06-13:
Hi @j4cbo , I am interested to buy 1 as well, however my email got blocked by your spam filter, is there an alternative way to communicate with you. I am in India.

kitor, 2024-06-14:
My riser arrived too, already installed and working!

### Some fun(?) facts i found in my setup.

Let's start with remark about PCIe topology in those devices:
- Both NVME slots go to PCH
- Factory x8 riser goes to CPU. Factory x4 riser goes to PCH
- j4cbo riser has pcie x4 slot on CPU, first (J2) m.2 on CPU, 2nd (J1) m.2 on PCH
- (not sure about WiFi slot, I don't use it atm)

I have:
- 2x NVME for boot devices (proxmox)
- 2x NVME for TrueNAS VM (pcie passthrough)
- 1x Aquantia AQC100 based 10G SFP+ NIC - Sonnet Solo 10G.

###
TL;DR:
- If you want to pass through a device to VM, use one of PCH slots to keep system going into low power C states
- If you need to pass through a device on bifurcated CPU slot, you likely need to pass through both devices.
- If you have Aquantia AQC100 based 10G NIC, there's a way to achieve C7 with very specific ASPM configuration

### PCIe passthrough and C states

While still running on standard riser with bifurcation card (and without 10G NIC) I found that system idles at C7, but the moment I boot TrueNAS VM it stops at C3. I had both passed through drivers at bifurcation card. As soon as I stop VM, system idles back at C7.

After some trial and error I found that for some reason passing through from bifurcated slot makes this happen. When I moved one of my storage SSDs to factory slot and passed only that drive - everything went to C6 (and even sometimes C7) just fine.

With new riser installed I confirmed this to be a case. I settled down by having one of boot and storage SSDs in factory slots, 2nd boot SSD in riser slot 1 (CPU) and 2nd storage ssd in riser slot 2 (PCH). This way BIOS still properly recognizes boot device in factory slot 1 and I pass through disks connected only to PCH slots.

### AQC100 card and ASPM / C states

 When card is plugged in, system reports max C2. Looking at lspci output gives this:
```
03:00.0 Ethernet controller: Aquantia Corp. Device 80b1 (rev 02)
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L0s L1, Exit Latency L0s unlimited, L1 unlimited
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
```

This is sus, as "unlimited" latency to my knowledge means once device enters state, it will never return from it.

Let's force ASPM enable on the card anyway:
```
root@pve:~# /opt/enable-aspm.sh 01:00.0 3
Device: 01:00.0 Ethernet controller: Aquantia Corp. Device 80b1 (rev 02)
	0x50 : 0x42 --> 0x43 ... [SUCCESS] L1 and L0s
```

Device goes briefly to C7... aaaand hard lock.
After some debugging I noticed in my boot scripts previously enabled L1 on PCIe port where this card is connected:
```
00:01.2 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x4) (rev 0a) (prog-if 00 [Normal decode])
		LnkCap:	Port #4, Speed 8GT/s, Width x4, ASPM L0s L1, Exit Latency L0s <256ns, L1 <8us
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
```

After disabling it (PCIe port ASPM disabled, Card ASPM L1 Enabled)... system works fine and goes to C7 again?!
I double checked that on stock M720 with factory riser - same happens.
I don't know wh it works but it works. Everything is running stable now.

### PCIe passthrough on bifurcated CPU slot (or on PCIe x4 / m.2 J2 on this riser)

After I received the riser, during my initial tests I found something odd about PCIe passthrough of bifurcated slot. On the old riser I had both drives in that slot. With the new riser NIC and one SSD goes to that slot.

If I passed through this particular SSD, as soon as I started VM, NIC went down. I tested pass through for other slots and just this one had that issue.

j4cbo, 2024-06-14:
Thanks for all the info! In my own system I'm currently running TrueNAS SCALE on bare metal, but I had been thinking of putting a Proxmox layer underneath. (TrueNAS works well enough as a VM host but Proxmox is much more powerful.) It's interesting that there's an interaction between passthrough and bifurcation, not terribly surprising but it seems like it "should" work.

I'm now sold out of the first batch of 20 boards! I'll ask the manufacturer about pricing for a follow-up batch. Anyone else who's interested in buying one, please let me know.

Go0oSer, 2024-06-19:
Are these still available?  Id be interested in 2 if so.

j4cbo, 2024-06-20:
Update: I've ordered more boards! :) I'll reach out to everyone who's contacted me once they're ready to ship.

joseal, 2024-06-24:
Hi @j4cbo, please count with me for one more board to EU. If there is anyone else interested in EU, I don't mind to join for a group buy.

thank you!

marcojk, 2024-06-24:
> This is the option i was referring to. Collect the order, pay to j4cbo, order shipped directly to europe and from here we will manage somehow :)

I'd be interested in 2 boards, and I still believe this could me an interesting alternative...

AndrewL82, 2024-06-26:
Hey, I'd be interested in buying 2 if shipping to Australia isn't too much :)

su-thomas, 2024-06-27:
> Hey, I'd be interested in buying 2 if shipping to Australia isn't too much :)

@AndrewL82 I'm also interested in Australia, perhaps we can bundle shipping to save on costs

randlor, 2024-07-28:
Hi, in the UK and interested in one. Thanks.

wouterbouvy, 2024-08-02:
Awesome work @j4cbo! I would be interested in buying 3 or 6 of your risers once your new batch arrives. Shipment to the EU.

[9 more replies]

closed 2025-01-18 by j4cbo