j4cbo/tiny5-m2-riser #3: Adapt this mod to M910X

systemerror-codenotfound, 2024-04-08:
The diagram for M910X should be similar to this, this is from a M710Q as i couldn't find one for the M910X
https://images.boosty.to/image/e28a63ee-ad04-4562-828a-b6a58697e4db?change_time=1674190987&croped=1&mh=625&mw=575

And this is the one for the M920Q
https://cxema.su/_sh/37/3745.jpg

and the Q270 in m910x chipset support the same bifurcation as the M920q

https://www.intel.la/content/www/xl/es/products/sku/98088/intel-q270-chipset/specifications.html

1x16 or 2x8 or 1x8+2x4

If i'n not wrong, right now you are getting 3 x4 pci ports from the actual PCI, 2 by bifurcating the PCI in 2 x4 and one x4 from the chipset
As per other comment in the @badger707 repo the pads are almost exactly the same and somebody actually tried to bifurcate it and it appear as 2 x4 lines and 1 x4 line, so the bifurcation on the CPU side works

j4cbo, 2024-04-09:
So, the main difference between my mod and https://github.com/badger707/m920q-pcie-bifurcation is that the M720/M920 series route an additional 4 lanes from the PCH to the riser slot, in addition to the 8 lanes from the CPU. Looking at the schematics for the IQ270IH1 (m710/m910/P320), it looks like the pinout is different and the extra 4 lanes aren't there:

[image]

I think https://github.com/badger707/m920q-pcie-bifurcation/issues/3 is the issue describing what you're looking for?

systemerror-codenotfound, 2024-04-26:
Okay, 
it's working on the m910x, i have to make some tests but i'm able to use both nvme's

The weird part is that i'm seeing another x8 pci bus

zzs95, 2024-04-26:
> the original comment was made by @zzs95

> it's working on the m910x, i have to make some tests but i'm able to use both nvme's

Thank you very much for your verification. I wondered if there was something wrong with my machine, so that I didn't make it.

I am more inclined to think that there is no 4x pcie from PCH on the m910x machine, because there is no 4x riser in the [accessory library](https://pcsupport.lenovo.com/us/en/products/desktops-and-all-in-ones/thinkcentre-m-series-desktops/thinkcentre-m910x/10my/parts/display/compatible?linkTrack=Caps%3ABody_SearchProduct&searchType=4&keyWordSearch=M910x%20Desktop%20%28ThinkCentre%29%20-%20Type%2010MY) Later models m920x and p340 tiny (m90q gen1) have this accessory.

m920x (01AJ929): 
[image]

m90q gen1(5C50W00876): 
[image]

j4cbo, 2024-04-26:
I think this discussion probably belongs over on https://github.com/badger707/m920q-pcie-bifurcation ? The riser I designed to allow putting three separate PCIe devices (two M.2s and an x4 slot) isn't possible because there's no routing from the PCH to the expansion slot - but bifurcation _does_ work on the 910 series.

galvesribeiro, 2024-04-26:
> Okay, it's working on the m910x, i have to make some tests but i'm able to use both nvme's
> 
> The weird part is that i'm seeing another x8 pci bus

After doing the PCIe bifurcation mode, add the "official" riser, it got me an extra x8 controller as well (x16, x8, x4):

```
[root@yoda:~] lspci
0000:00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers 
0000:00:01.0 PCI bridge: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) 
0000:00:01.1 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x8) 
0000:00:01.2 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x4) 
0000:00:02.0 VGA compatible controller: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] 
0000:00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model 
0000:00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller 
0000:00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM 
0000:00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller 
0000:00:16.3 Serial controller: Intel Corporation Cannon Lake PCH Active Management Technology - SOL 
0000:00:17.0 SATA controller: Intel Corporation Cannon Lake PCH-H AHCI Controller 
0000:00:1b.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 
0000:00:1b.4 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #21 
0000:00:1c.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #6 
0000:00:1f.0 ISA bridge: Intel Corporation Q370 Chipset LPC/eSPI Controller 
0000:00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS 
0000:00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller 
0000:00:1f.5 Serial bus controller: Intel Corporation Cannon Lake PCH SPI Controller 
0000:00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (7) I219-LM 
0000:02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO 
0000:03:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO 
0000:04:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO 
0000:05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO 
0000:06:00.0 Non-Volatile memory controller: Phison Electronics Corporation PS5013 E13 NVMe Controller 
```

With that, I can put the two NVMe M.2 there and it just work. As you can see on `lspci` I have 4 Samsung 980 PRO there and they all work just fine (The Phison is a 2230 M.2 NVMe on the Wifi slot).

With @j4cbo Riser I will be able to use the other extra 4 lanes from the PCH and use along with the two M.2 a dual 10G NIC.

But yeah, you should go post on the other repo where you can get more help about the PCIe mod.

closed 2024-05-03 by systemerror-codenotfound