j4cbo/tiny5-m2-riser #4: [Suggestion] Use PCIe x16 connector

galvesribeiro, 2024-04-15:
On the prototype picture I see a x4 connector for the PCIe card.

Would it be possible to use a x16 connector with only x4 wired? The reason I'm asking is so we can easily use bigger cards using the same riser. 

I know we can use an open-ended PCIe connector but those are harder to find for manufacture as they are not safer because leave the card connections "out".

I hope you consider the suggestion.

Thanks!

j4cbo, 2024-04-15:
An x16 connector won't fit because the space where the pins would go is needed for the signal traces to the M.2 connectors. However, I think I can source an open-back connector...

galvesribeiro, 2024-04-15:
Ok. Duck tape on the open connectors it is 😄

j4cbo, 2024-04-15:
If you want it to look nice and professional, pick up some Kapton tape :)

closed 2024-04-15 by galvesribeiro