j4cbo/tiny5-m2-riser #6: Showcase riser working

galvesribeiro, 2024-07-14:
I just wanna share with others that I have full success using this riser.

I have 4x Samsung 980 Pro 2TB (two in the riser itself and two in the back), one M.2 on the Wifi slot (Phison on the screenshot) and a dual 25Gb Mellanox Connect-X 4 NIC, and another SATA SSD.

[image]

Thanks @j4cbo for the amazing work!

galvesribeiro, 2024-10-15:
> @galvesribeiro, were you get the full bandwidth for the dual 25GB with all NVMe in use?

Good question. I tested with only 1x 25GB connected and I got full bandwidth but never tried saturate both. Right now the machine is shutdown as I moved and have reassembled my lab yet. I'll do the test later on and report back.

ilium007, 2025-07-15:
> I just wanna share with others that I have full success using this riser.
>
> I have 4x Samsung 980 Pro 2TB (two in the riser itself and two in the back), one M.2 on the Wifi slot (Phison on the [...]

Have you git any pics of how this is all fitted inside the case?

StancuFlorin, 2026-01-08:
Hi @galvesribeiro 

Can you add any pics with with the wifi slot and the other sata SSD?

Thanks

ilium007, 2026-11-11:
My wifi slot nvme stopped worked after doing the bifurcation resistor mod