j4cbo/tiny5-m2-riser #7: Passthrough compatible version

fajeczek00, 2024-07-19:
As stated in the group buy when passing through only one of the bifrucated ports on proxmox the host doesn't enter proper c-states.
Would it be possible to create a PCB revision with all 8 bifrucated lanes going into 2 m.2 ports and PCH lanes going into the regular X4 slot ?

In that scenario if bios allows the bifrucated slots to boot it would be possible to use 2 bifrucated slots for a RAID 1 proxmox installation mitigating  passthrough sleep issues, leaving all the other slots to be used for various passthrough configs

j4cbo, 2024-08-10:
In theory, yes. The issue is geometry - the X4 slot is right up against the card edge connector, so there's nowhere to put the traces. Rearranging the connections that way would require going to a 6 or 8 layer PCB, adding significant cost (and requiring me to get different PCB layout software). So it's not on my to-do list, sorry :(

wrobelda, 2025-06-10:
Well, I just reported another issue closely related to to what @fajeczek00 suggests: https://github.com/j4cbo/tiny5-m2-riser/issues/19