j4cbo/tiny5-m2-riser #8: Any PCB left?

Jatsekku, 2024-10-01:
@j4cbo Amazing project!
Do you have any PCBs left or planing another round of ordering?
I'd be interested in one ;)

asche77, 2024-10-01:
If there is EU shipping, interested as well...

j4cbo, 2024-10-02:
I have 3 more assembled units available (and a bunch of bare PCBs, but that would be a big soldering project) - send me an email, my address is on my Github profile. I'm asking $55 USD each, and shipping is about $20 to most of the world or $5 within the US.

dionorgua, 2024-10-02:
I also want to get one in EU (Poland).

j4cbo, 2024-10-05:
Sorry, I'm all out now. I'm working on a new batch and I'll get back to everyone who's emailed me when I have them in stock again.

j4cbo, 2024-11-01:
I just received a new batch. If I haven't replied to your email please ping me again :)

sherbibv, 2024-11-17:
I'm interested, but I am not sure if I need the bifurcation mod even if I don't use the other pcie slot. I would like it just for the extra 2 m.2 ssds.

snzsnz, 2025-03-20:
Are you still selling these? Price shipped to NZ?

j4cbo, 2025-03-20:
I'm all out, but user evil2k2 is doing a build in Germany: https://github.com/j4cbo/tiny5-m2-riser/issues/11

closed 2025-03-20 by j4cbo

mr4lexndr, 2025-08-13:
Hi All, 

I have made a **batch of boards**. 
📟  They are **available [here](https://www.tindie.com/products/letbco/pcie-riser-2x-nvme-x4-for-lenovo-tiny-m920m720/)** 
📦 I ship them from the UK and the EU. 

🔩 As an add-on, you can also obtain all the necessary **fasteners and brackets**.