j4cbo/tiny5-m2-riser #10: Is this plug & play?

phere-x, 2025-01-20:
I'm new to this side of Tiny's entirely so forgive me for the novice questions. 

I'm assuming it requires no soldering on the board or mods done by me? I came from https://github.com/a-little-wifi/Tinyriser first and it mentioned it should work by just plugging it in, is it the same for this? This seems the same, save for a nifty fan header it has but there's no instructions for ordering your own like the awesome guide j4cbo wrote out for us https://github.com/j4cbo/tiny5-m2-riser/tree/main/pcb-v2

Or if anyone's getting/had it done in the UK, anyone willing to help others out?

chocmake, 2025-01-21:
> I'm assuming it requires no soldering on the board or mods done by me?
Per the readme it requires soldering two 0402 resistors, on the underside of the motherboard. The riser by a-little-wifi's was unique in that it no mods were required.

j4cbo, 2025-01-21:
Yep, exactly.

The motherboard header has two separate PCIe interfaces: 8 lanes to the CPU, and 4 lanes to the PCH. Lenovo's 01AJ940 uses the CPU interface, and the 01AJ929 uses the PCH interface. I'm not sure why they did it this way but I suspect that the Thunderbolt addon has to be connected to the PCH, while the CPU interface is higher-bandwidth and better for a GPU.

The riser by a-little-wifi routes the x8 CPU interface directly to a PCIe slot, and the x4 PCH interface to an M.2 connector. Mine  instead splits the x8 CPU interface, routing half of it to an x4 PCIe slot and the other half to a second M.2 (and lane-reverses both of them). That allows for an extra SSD while still being able to fit a 10G Ethernet card in the x4 slot. But it also means bifurcating the x8 PCIe link to the CPU, which is the part that requires soldering and doesn't work with the M720q.

phere-x, 2025-01-26:
Really appreciate the responses & clarifications! I was really eager to try it out but soldering skills let alone the equipment is something I don't have to be attempting badger's soldering bifurcation.

May be a lot to ask, but any idea of how to get a hold of my own a-little-wifi's part? I only ever came across this from [tindie](https://www.tindie.com/products/crimier/tinyriser-a-better-lenovo-tiny-pc-riser/) but the seller is away & some poor reviews here n there. I've never used this site either. Not sure if anyone's doing anything on forum's but I couldn't find anything using google & some keywords.

evil2k2, 2025-01-26:
> Really appreciate the responses & clarifications! I was really eager to try it out but soldering skills let alone the equipment is something I don't have to be attempting badger's soldering [...]

For soldering the resistors you can ask my friend @AcidzDev (Discord acidz) also UK.
For the riser from @a-little-wifi's ask him on his repo.

maldworth, 2025-03-08:
> That allows for an extra SSD while still being able to fit a 10G Ethernet card in the x4 slot

If I have a m720q, could I use your riser, and I don't bother soldering the 2 resistors (because bifurcation doesn't work on m720q)? Then I would just get one of the nvme slots working, and then I'd get the pcie slot with just x4? So essentially a solderless option?

j4cbo, 2025-03-10:
The issue is the lane-reversal part. The Intel CPU in these systems actually has a 16-lane PCIe interface, but the Lenovo motherboard only routes lanes 0-7 to the riser. The CPU allows the port to be set up as x16 (lanes 0-15), x8+x8 (lanes 0-7 + 8-15), or x8+x4+x4 (lanes 0-7, 8-11, and 12-15). Fortunately there's also a lane-reversal feature in the CPU, which swaps the order of the lanes, _including_ how they're bifurcated - enabling both bifurcation and lane reversal means lanes 3-0 (in reverse order) form one logical port, and lanes 7-4 (in reverse order) form a second logical port. My riser then routes the traces so as to swap the order back.

That said... there's an optional feature of the PCIe spec that allows devices to implement automatic lane reversal. So it might work! My Aquantia-based network card seemed to come up in this situation on an M920x without the resistors soldered, though I don't remember what the link width was. But not all devices will work, and it's usually not well-documented whether or not reversal is supported.

Ryanstanley4, 2025-05-07:
If this is the case I think the [README.md](https://github.com/j4cbo/tiny5-m2-riser/blob/main/README.md), should be updated as it reads “_for Lenovo Tiny5 series (M720q/M920q/M920x, etc)”_

I have bought several of these cards for the m720q based on this, it looks like I need to make a new plan.