j4cbo/tiny5-m2-riser #11: Orders from EU

CaneTLOTW, 2025-02-04:
Hi, are there any plans to make them available from Europe?

evil2k2, 2025-02-06:
Hey there
I'm currently having 50 pieces produced. But I will assemble them by hand. But it will be a while before I can send out the first ones. I would report here when I'm ready. I come from Germany so EU shipping is no problem.

CaneTLOTW, 2025-02-06:
Das macht den Versand natürlich einfacher ;)

Feel free to add me to the list of interested people with 2 pieces.
Do you already have an idea about pricing?

evil2k2, 2025-02-06:
hehe ja das stimmt natürlich 😉

No not at the moment i'm searching the parts for the project.

lmbacelar, 2025-02-09:
Hi!
I would also be interested in having 2 or 3 of these.
Please let me know when they are available.

devnow, 2025-02-11:
Hi
I would be interested in two riser boards. Shipped to Switzerland or Germany.

Thanks

vgchl, 2025-02-23:
Hi, I'm interested in two boards as well. Shipping to Singapore.

kursatozkan, 2025-03-01:
I am also interested for 4 pcs please. Thanks a lot! Regards,

moe9301, 2025-03-03:
Hi, interested for 2 pcs shipped to Germany :) Nice work

vgchl, 2025-03-03:
@evil2k2 Any estimates on when you'll be ready to ship some out?

evil2k2, 2025-03-03:
I finally got all the parts, only solder paste still missing atm.
But I think I can try to build the first one together these days.

Did no accurate calculation for the price but It should be around 50€.

Loloke, 2025-03-07:
i also interested for 1-2 pcs, shipped to hungary.

j4cbo, 2025-03-10:
@evil2k2 I'd suggest not placing the C5 capacitor - huge thanks to @devnow for noticing the issue with the design in issue #15.

The published version with that capacitor does work (and it's what I've been using in my own server for a few months) but will likely have very poor load transient response, which could lead to system instability.

gastrek, 2025-03-14:
I am interested in 1 PCB as well.

evil2k2, 2025-03-15:
I just finished the first one, tomorrow I'll test it and if everything works I'll do the others.
The ASA filament for the holders will arrive next week and the screws should arrive soon too.

[image]
[image]
[image]

joseal, 2025-03-16:
I am also interested in 1 PCB, could you please confirm the price (without the 3d printed parts)?

zididadaday, 2025-03-16:
Hi! Cool that you take the initiative. Can I be added for 2 pieces as well?

AnisseH, 2025-03-18:
Hello
i would be interested for 3 pieces without 3d brackets

otopetrik, 2025-03-20:
Hello,
I am also interested in 1 piece (shipped to Czech Republic).
Thanks!

DavidCarl, 2025-03-21:
Could also be interested in one or two shipped to Danmark :)

Thank you!

zididadaday, 2025-03-21:
> I just finished the first one, tomorrow I'll test it and if everything works I'll do the others. The ASA filament for the holders will arrive next week and the screws should arrive soon too.
> 

Finally weekend - how has the testing gone - 💯? @evil2k2

evil2k2, 2025-03-21:
> > I just finished the first one, tomorrow I'll test it and if everything works I'll do the others. The ASA filament for the holders will arrive next week and the screws should arrive soon too.
> 
> Finally weekend - how has the testing gone - 💯? [@evil2k2](https://github.com/evil2k2)

All fine works like expected, have 5 done yet.
Screws, esd bags and packaging is also arrived.
But I don't have time rn have a few consoles for repair atm.
3d printed parts are pla+ until I figure out how to print asa right 😅

Would be great if any of you wrote me a mail, but please from first to last.
Write these please also in the mail.
1, 2 and so one
I need the full address, phone number and mail address for shipping outside from Germany.
I use probably UPS for shipping, but I don't have a price for EU shipping.
Outside of EU I must check the shipping possibilities to say a price.

Adresse please like these:
Name:
Adress:
Postcode:
City:
Country:

The price is:
45€ without the 3d printed parts and screws 
50€ with these parts and assembled 
Payment is only bank transfer (I don't like PayPal)
My mail is: evil.santa.e4s@gmail.com
Will start shipping probably middle next week.

[image]

PerMejdal, 2025-03-21:
> I just finished the first one, tomorrow I'll test it and if everything works I'll do the others. The ASA filament for the holders will arrive next week and the screws should arrive soon too.

Why print i ASA? It is flamable, and hard to print with.
PETG is not flamable, but still mechanically strong.

evil2k2, 2025-03-21:
I was thinking about ASA because it can withstand high temperatures. But you're right, I should buy PETG instead.
I have just started with 3D printing and have no experience with it yet.

dafula, 2025-03-30:
Hi, I'm interested in 1 piece with 3D printed parts. Shipping to Germany. Is there anything left?

wrobelda, 2025-06-01:
Any update here? Need a riser only, sans 3d parts.

kursatozkan, 2025-06-03:
I was also waiting for these risers without the 3d part. Is it possible to ship two to Bulgaria? Thanks in advance.

gelucrisan, 2025-06-09:
I need two as well, I am located in Romania.
@evil2k2 I was not able to find your email address, can you  please share so I can send you the shipping info?
Thank you

Clarion30, 2025-07-03:
Hallo, hatte schon mal eine email geschrieben, hätte gerne auch eine Karte.

mr4lexndr, 2025-08-13:
Hi All, 

I have made a **batch of boards**. 
📟  They are **available [here](https://www.tindie.com/products/letbco/pcie-riser-2x-nvme-x4-for-lenovo-tiny-m920m720/)** 
📦 I ship them from the UK and the EU. 

🔩 As an add-on, you can also obtain all the necessary **fasteners and brackets**.

wrobelda, 2025-10-01:
@Piorek99 did you solder the SMD elements on the motherboard as instructed?

mr4lexndr, 2025-10-01:
> [@Piorek99](https://github.com/Piorek99) did you solder the SMD elements on the motherboard as instructed?

@Piorek99, this would be the most likely cause of the problem. J1 uses PCIe lanes from PCH, which should work regardless of bifurcation. J2 will not work, and J3 might work, but I haven't tested it. 

Bifurcation is done by soldering two SMD resistors at the bottom of the motherboard, as explained in this repo.

Janzzze, 2026-10-15:
> 31\. > Hi there,
>          > Anyone still selling these in Europe? I would like to buy 2-3 units.
>          > Thanks!
>          
>          
>          Hi,
>          I have them available [here](https://www.tindie.com/products/letbco/risr5-v3-dual-nvme-x4-pcie-lenovo-m920m720p330/) ✅

Are you planning to sell these again? have been out of stock for a while.

dpratt-ops, 2026-10-15:
> 31\. > 31. > Hi there,
>      > > Anyone still selling these in Europe? I would like to buy 2-3 units.
>      > > Thanks!
>      > ```
>      >      Hi,
>      >      I have them available [here](https://www.tindie.com/products/letbco/risr5-v3-dual-nvme-x4-pcie-lenovo-m920m720p330/) ✅
>      > ```
>      
>      
>      Are you planning to sell these again? have been out of stock for a while.

I just ordered 5 of these. If we can figure out shipping logistics and import stuff, I'll sell them.

JuniorJPDJ, 2026-10-15:
let us know please, I'd buy one

Janzzze, 2026-10-15:
I'd buy one as well.

bullii84, 2026-10-16:
I also want to buy one. Please let me know when it becomes available.

dpratt-ops, 2026-10-16:
@JuniorJPDJ  @Janzzze  @bullii84 
I've received the 5 unites I ordered. I will be testing them to confirm they work. Please email me at davidwpratt@gmail.com your contact and address information and we can discuss price and logistics for shipping depending on where you're located.

mr4lexndr, 2026-10-17:
> 26\. > 31. > Hi there,
>      > > Anyone still selling these in Europe? I would like to buy 2-3 units.
>      > > Thanks!
>      > ```
>      >      Hi,
>      >      I have them available [here](https://www.tindie.com/products/letbco/risr5-v3-dual-nvme-x4-pcie-lenovo-m920m720p330/) ✅
>      > ```
>      
>      
>      Are you planning to sell these again? have been out of stock for a while.
@Janzzze @JuniorJPDJ @bullii84  

🎉 **They're available now!** You can order it here 👉 **[letb.io/risr5](https://letb.io/risr5)**

🚚 Ships worldwide · 🔧 Optional hardware kit

dpratt-ops, 2026-10-20:
I have qty 4 available in black [for order](https://www.tindie.com/products/dp_products/2x-nvme-x4-pcie-riser-for-p330m920xm720/), but shipping is from US. I know this isn't ideal for EU or other customers but I'm willing to ship internationally and work on VATs and Custom fees as needed.