j4cbo/tiny5-m2-riser #12: Can it be combined with TinyRiser?

CaneTLOTW, 2025-02-06:
Could this Riser also add the function to use the additional 4x connection, like here?
https://www.tindie.com/products/crimier/tinyriser-a-better-lenovo-tiny-pc-riser/

[image]
Made by https://twitter.com/WifiCable_

> This is possible because the riser slot on the motherboard actually contains two PCIe links, one x8 link and one x4 link. The x8 link comes from the CPU and was meant for GPU equipped models, while the x4 link comes from the PCH and was meant for the Thunderbolt and NIC equipped models. Lenovo made two different risers for these configurations, only using one link at a time, but this riser gives you both PCIe links at once! The two links are completely independent and are able to function at the same time.

evil2k2, 2025-02-06:
Check the repo https://github.com/a-little-wifi/Tinyriser

CaneTLOTW, 2025-02-06:
I should read the full documentation 1st.
It is already included...

https://github.com/j4cbo/tiny5-m2-riser/blob/main/images/block-diagram.png