j4cbo/tiny5-m2-riser #13: Booting from SSD connected via riser

derpicknicker1, 2025-02-07:
I tried to boot from a ssd connected via the riser. But I can not see the ssd in the boot menu. Tried both slots of the riser. If I use the onboard slots, it works like expected. Is this a known 'issue' or am I doing something wrong?

dionorgua, 2025-02-07:
System BIOS is pretty limited. Even if you plug SSD to wifi slot (via adapter) system will not detect it as bootable device.
At the same time system will boot  from it if there is no more options available

evil2k2, 2025-02-07:
You need to install your operating system to this drive.
Also be sure you don't have other systems installed also remove all drives from boot order in bios.
This works also with ssd connected to WiFi Slot.

j4cbo, 2025-02-09:
Yep, as @dionorgua said, this is expected. SSDs connected to anything besides the on-motherboard M.2 slots will not appear in the BIOS or the boot menu, so you'll need to use one of them (or SATA) for your boot drive. My understanding is that if there are no other bootable devices, the BIOS will be able to boot from drives on the riser even though they don't appear in the menu, but I haven't tried it myself.

closed 2025-02-09 by j4cbo

dionorgua, 2025-02-10:
My m920x is booting from 2242 SSD connected to 'wifi' slot. This is my 'production' setup.

[image]

It works but you need to make sure you don't have anything bootable on 'supported' drives (two standard m.2 SSDs or SATA caddy)

ilium007, 2025-07-17:
> My m920x is booting from 2242 SSD connected to 'wifi' slot. This is my 'production' setup.
> 
> [image]
> 
> It works but you need to make sure you don't have anything bootable on 'supported' drives (two standard m.2 SSDs or SATA caddy)

Nice setup. Do you have a link to the M.2-E to nVME adaptor?