j4cbo/tiny5-m2-riser #14: Do you think a riser for 12x or 8x nvme archivable ?

ersandemirayak, 2025-02-10:
3x or 2x  asm1184 to extend the number of nvme to 12, for a SSD NAS card?

[image]

CaneTLOTW, 2025-02-17:
Some KI based remarks ;)

Below is a high-level explanation of what’s going on with the ASMedia ASM1184 switch (and similar chips in the ASMedia product line) and why you’ll often see advice pointing toward a “real” PCIe switch (like Broadcom/PLX) when trying to fan out to many NVMe devices.

---

## 1. What is the ASM1184?

- **ASM1184** is a relatively small PCIe packet switch that typically provides 1 upstream PCIe lane and up to 4 downstream ports (often each is a single-lane link as well).
- It is usually marketed for adding multiple USB3 controllers, small M.2 (WiFi/SSD) slots, or other low-bandwidth devices behind a single x1 or x2 link.
- You’ll sometimes see it on “quad M.2 adapters” that each slot is forced to use PCIe x1, or on riser boards to attach multiple small devices to a single host port.

Because each downstream port in many ASM1184-based designs is only **PCIe x1**, each NVMe SSD—if attached directly—would be limited to x1 bandwidth. That can be enough for embedded or low-end uses (still hundreds of MB/s per SSD), but it does *not* tap the full performance typical NVMe drives can deliver at x4.

---

## 2. Can I chain or combine multiple ASM1184 chips to get 8 or 12 NVMe drives?

In principle, yes:
- You could place two or three ASM1184 chips on a single board, each connected to the upstream slot (or chained from each other).  
- This would give you 2× or 3× “4-port groups,” so 8 or 12 potential downstream ports in total.

However, there are significant design and performance caveats:

1. **Upstream Lane Width** – The ASM1184 generally expects a single-lane or dual-lane upstream (depending on variant). Even if you feed 2 or 3 ASM1184 chips from (for example) a PCIe x4 slot, you still have to subdivide that x4 into multiple x1 or x2 links per chip. As soon as you start slicing up a single x4 link into multiple x1 “branches,” each group of 4 M.2 slots behind an ASM1184 is contending for limited upstream bandwidth. This can severely limit the total performance if all drives are active.

2. **Chaining Complexity** – You can chain small PCIe switches (one ASM1184 attached upstream, with others downstream), but the link negotiations and bridging can get tricky. It can become finicky or unstable unless carefully designed—and you still share the same limited total bandwidth.

3. **Firmware/Compatibility** – Some motherboards or BIOS/UEFI firmwares are pickier than others about enumerating multiple chained PCIe bridges. You may encounter boards that simply won’t show all the drives or that get stuck at boot, requiring advanced firmware patches or luck.

Because of these constraints, you do not commonly see “3× ASM1184 for 12 M.2 NVMe” or “2× ASM1184 for 8 M.2 NVMe” as a mainstream product. It *can* be done, but:

- You often end up saturating a single-lane (or x2) upstream link, making the performance for each drive quite low.
- The card design becomes more complex; at that point, it’s usually more cost- and performance-effective to switch to a bigger, full-fledged PCIe switch.

---

## 3. Using a “Real” PCIe Switch (PLX/Broadcom, Microchip, or ASMedia ASM28xx)

When you see high-end “8-port” or “16-port” NVMe carrier boards, they typically use a larger PCIe switch, such as:

- **Broadcom (PLX) PEX series** (e.g., PEX 8718, 8732, 8747, etc.)
- **Microchip Switchtec** series
- **ASMedia ASM2812 / ASM2824** (supporting more lanes/ports and higher bandwidth)

These chips:
- Accept a wider upstream interface (e.g., PCIe x8 or x16).
- Provide multiple *x4 or x2* downstream ports, so you can run each NVMe at near its native PCIe 3.0 or 4.0 speeds.
- Have more mature firmware for enumerating a large number of endpoints behind one switch.

For an **SSD NAS card** (implying you truly want to push multi-gigabyte per second read/write performance across multiple M.2 drives), you really want an actual multi-lane upstream and a proper switch chip that can feed each drive with enough bandwidth. That’s the difference between a “fun experiment with lots of x1 drives behind multiple cheap switches” and a stable, high-performance storage solution.

---

## 4. Conclusion

- **3× or 2× ASM1184** to get 12 or 8 M.2 slots is *technically* possible, but generally not recommended for true NVMe performance because you will be severely bottlenecked by the upstream lane(s), plus the design/compatibility headaches of chaining multiple small switches.
- If your use-case only needs each drive at ~500–900 MB/s (PCIe 3.0 x1 speeds) for lower-end “NAS” throughput, then you could explore it. Just be aware it’s outside the typical reference design, and you may run into motherboard/BIOS quirks.
- For a robust multi-SSD NAS solution at full NVMe speeds, look at a larger (and unfortunately more expensive) PCIe switch solution—e.g., a **PLX/Broadcom** or **ASM28xx**-based board—which can properly fan out x8 or x16 upstream to multiple x4 M.2 ports. That’s how commercial 8+ port NVMe HBAs or storage boards do it.

In short, “yes, you *could* do 2× or 3× ASM1184,” but “no, it usually doesn’t give you the kind of performance or stability you expect from a proper 8–12 port NVMe card.” If you need a serious SSD NAS card, a bigger switch chip is the way to go.

j4cbo, 2025-02-17:
I'd be worried about power supply and thermal limits... plus, fitting PCIe switches on the back of the riser with connectors on the front would likely require switching to an expensive HDI PCB process. In short, not on my to-do list :)

closed 2025-03-13 by j4cbo

evil2k2, 2025-03-16:
There are two cards available with integrated PCIe switching Chip.

[image]

[image]

https://a.aliexpress.com/_Ev5BUx2
4x NVMe 

Or this beast 
8x NVMe with X16 connector (works down to X4)

[image]

[image]

Both are fitting more or less in an Lenovo Tiny 5 (m720, m910x, m920)