j4cbo/tiny5-m2-riser #15: buck converter v2 remarks

devnow, 2025-02-11:
Thanks for releasing the designs. Remark on the buck converter feedback loop [3], the datasheet is unusually sparse on information.
The data sheet recommends (Table 8-2. Recommended Component Values [1]) a R6=20k, R5=90k (or 91k for E24 values) for 3.3V and a feed forward capacitor C_FF=C5 between 0-100pF. The reference boars use 22pF [2]. Output Voltage Setpoint V_OUT = 0.6 * (1+ R_upper/R_lower) [2]
I have no experience in NV drives current profiles. Judging from your v2 design notes it's not your first pci-e design rodeo, glad to learn for further insights.

[1] [TPS51386 data sheet](https://www.ti.com/lit/ds/symlink/tps51386.pdf)
[2] [User's Guide TPS51386 Step-Down Converter Evaluation Module](https://www.ti.com/lit/ug/sluucn0b/sluucn0b.pdf)
[3] [Design considerations for a resistive feedback divider in a DC/DC converter](https://www.ti.com/lit/an/slyt469/slyt469.pdf)

j4cbo, 2025-03-10:
Good catch! The feedback resistors were chosen to give an acceptable output voltage while staying within the E12 series, but the capacitor is just a copy-paste error. 😬 Props to TI for building a control loop that works with such excessive feedforward capacitance.

It's interesting that the EVM includes a capacitor at all there... the datasheet notes that the regulator "does not require external compensation" and the reference schematic on page 1 doesn't include one, so I'm going to suggest just removing C5 instead. I'll send an email to everyone who's purchased a V2 board and update the schematics in git.

Fironet, 2025-03-15:
> so I'm going to suggest just removing C5 instead

Just to confirm, you're suggesting users desolder the component circled here?

[image]

crimeofheart, 2025-06-08:
> Good catch! The feedback resistors were chosen to give an acceptable output voltage while staying within the E12 series, but the capacitor is just a copy-paste error. 😬 Props to TI for building a [...]

I guess v1 owners are safe?