j4cbo/tiny5-m2-riser #18: 配置电阻是2个还是3个？

xianxixi, 2025-05-16:
[image]
根据这张图，配置电阻不应该有3个吗？cfg2,cfg5,cfg6

j4cbo, 2025-05-16:
CFG5 (R888) is already set at the factory, so only the resistors for CFG2 (R885) and CFG6 (R889) need to be added:
CFG5（R888）已在工厂设置，因此只需添加 CFG2（R885）和 CFG6（R889）的电阻：

[image]

kqkq, 2025-05-21:
Don't ever try to replicate this design in China. The inductor and M.2 connectors are insanely expensive. You may want a redesign and use parts easily available in China. 不要尝试在国内复刻这个设计，他选的电感、电源芯片、M.2连接器都贵的离谱。最好是重新设计并选用国内容易买到的零件。

j4cbo, 2025-05-21:
Those vertical M.2 connectors are hard to find. If there's a cheaper option, I'd like to hear about it :) The inductor can be substituted too.

kqkq, 2025-05-21:
> Those vertical M.2 connectors are hard to find. If there's a cheaper option, I'd like to hear about it :) The inductor can be substituted too.

regarding the M.2 connector, could you consider to change to:

Brand: LOTES
Model: APCI0110-P001A
JLC number: C841669

this only cost 1/10 of the current connector.

j4cbo, 2025-05-21:
That connector puts the M.2 card parallel to the riser, but it needs to be sticking out at a 90 degree angle :(

kqkq, 2025-05-22:
> That connector puts the M.2 card parallel to the riser, but it needs to be sticking out at a 90 degree angle :( 

APCI011-P001-A is the correct model, but the LCSC number is incorrect... My bad...

I know it needs to be vertical. However, this model is not available at LCSC. Can you check AliExpress or other websites?

Thank you.

j4cbo, 2025-06-01:
Do you know where to find a datasheet for any of these? It looks like many are available from various sellers on Aliexpress, but I can't find a dimensioned drawing anywhere. They might be compatible with the same layout but it'd be good to be able to check...

kqkq, 2025-06-03:
> Do you know where to find a datasheet for any of these? It looks like many are available from various sellers on Aliexpress, but I can't find a dimensioned drawing anywhere. They might be compatible with the same layout but it'd be good to be able to check...

I don't have the datasheet. I did purchase some to try, but the footprint is not compatible.