j4cbo/tiny5-m2-riser #20: How are you fitting the SATA boot SSD in the case?

ilium007, 2025-07-13:
With the additional nvme drives, PCIe card (assuming 10GB or 2.5GB NIC), how are you fitting the boot SSD via the SATA flex cable? Do you have any more pics on how you have it all fitting?

I have a Lenovo P330 Tiny I am considering doing this mod on - awesome project! If this works it will solve so many home lab problems for me!

My P330 has a large black heatsink plate under the PCIe card / SSD position:

[image]

[image]

[image]

ilium007, 2025-07-17:
I have purchased one of these as they are quite small - just need a 2230 SATA SSD.

[image]

j4cbo, 2025-07-17:
Interesting, my m920x doesn't have that heatspreader. Maybe it's specific to the P330 Tiny?

I used a SanDisk "Half Slim" SSD and designed an extra 3d-printed piece that goes underneath the m2-mount component with some arms that come off to hold the SSD. You could probably do something similar with your 2230 adapter. Here's how it fits together:

[image]

[image]

ctcDNightmare, 2025-09-09:
> Interesting, my m920x doesn't have that heatspreader. Maybe it's specific to the P330 Tiny?
> 
> I used a SanDisk "Half Slim" SSD and designed an extra 3d-printed piece that goes underneath the m2-mount component with some arms that come off to hold the SSD. You could probably do something similar with your 2230 adapter. Here's how it fits together:
> 
> [image]
> 
> <img alt="Image" width="593" height="303" src="https://private-user-images.githubusercontent.com/537158/467561753-7bf3ea66-c600-4374-92fa-48ca897e0458.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc0MDY2NDEsIm5iZiI6MTc1NzQwNjM0MSwicGF0aCI6Ii81MzcxNTgvNDY3NTYxNzUzLTdiZjNlYTY2LWM2MDAtNDM3NC05MmZhLTQ4Y2E4OTdlMDQ1OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkwOVQwODI1NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMTFkOTEzYTFjZWQ1Nzg1OTE0MWM3ZjNmZWQwZWRlZWEzNDI3YWVlMjU1Y2Q0YjYyZWRkODI0MmE0ZmZjOTQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZLc3-1mRD2l3ro-3HJ6H9cMdIVmp0cY5ZJrWljE44aI">

Interesting, my m920x also has that heatsink installed but I couldn't find it in the maintenance docs from Lenovo for either the p330 or the 920x 🤷 

Could you share the .stl of that additional adapter by any chance? Looked in the repo, but so far didn't see them or I'm blind

Trying to squeeze everything planned in the m920x is a challenge, your bracket sure will help.
(2x m.2, 1x sata, 1x sas 4e hba, wifi-slot > 2.5gb nic)

zollak, 2025-09-09:
Hi guys! It is an awesome project! Thanks for sharing this! Would you mind to share some info about the available speeds? I know it depends on many things, but knowing the nvme types and the measured bandwidth would be super welcome!