j4cbo/tiny5-m2-riser #24: The computer burned out with the board connected.

klaudiawlodarskaa-ui, 2025-08-09:
Hi, I ordered three boards from JLCPCB exactly according to the instructions. Nothing was modified. The first board worked for a while, but when I reconnected it, the computer burned out. There was smoke and a foul smell coming from the computer's ventilation grille near the power plug. The computer already had two M.2 drives connected. The second computer burned out the first time with the same configuration, the only difference being that the first time, the case was not closed while the computer was running. I don't know if this has any effect, but the computer has a case closure sensor. The third computer had a other board connected, without drives, and it also burned out the first time. Do you have any suggestions as to what might have happened? All the computers were m920x with different specifications.

j4cbo, 2025-08-12:
Yikes, that definitely shouldn't happen!

The oscillator provides the M.2 `SUSCLK` signal, which is supposed to be present in suspend states where I wasn't sure if 3V3 would be available. Most NVMe devices don't require it but some do. It might be safest to just remove the oscillator entirely?

Also, is there any chance the riser was connected or removed while the system was plugged in (even if powered down)?

mr4lexndr, 2025-08-12:
In my case (I burned two PCs), the riser was connected before the power was plugged in. However, I had boot enabled on power loss, so it turned on immediately after plugging in the supply. In one of them, it worked for the first time and then burned after the second plug-in.

The strange part is that 3V3_S5 is capable of 5A and the oscillator wouldn't draw more than 1.5mA. Is there any case that could directly cause this failure?

mr4lexndr, 2025-08-12:
I did further testing. 

After replacing the mentioned MOSFET, it was running fine without PCIE bifurcation (two cfg resistors not present), riser in the slot. With resistors in place and the riser in, it burned red straight away.

The next step was to replace the MOSFET again and test without the riser. All ok. 

Then I removed the oscillator from the riser (the only thing connected to the 3V3_S5 line of the MOSFET), put it in place, and the MOSFET glowed red again with a lot of smoke. 

Currently, I am out of ideas, as I cannot find any other relation of 3V3_S5 MOSFET to the riser or PCIe.

j4cbo, 2025-08-12:
I'm baffled too. I would expect that MOSFET to win a fight with just about anything on the riser board, and I really don't understand how disabling or enabling bifurcation would affect the outcome.

The FET glowing and smoking makes me suspect a hard short to ground. Maybe there's intermittent contact and removing/reinserting the riser changes the alignment enough to make a difference? It's a long shot but it's all I've got.

crimeofheart, 2025-08-12:
@mr4lexndr Can you upload a photo showing how exactly you did it? I already replaced mosfets twice and want this drama to end already :) 

Thanks in advance!

mr4lexndr, 2025-08-12:
The first photo shows an incorrect, although possible, way to insert the riser. Pay close attention to the alignment of pads with pins in the socket. The second photo is correct.

The next two show which pads I cut out with a scalpel for testing. This is unnecessary now (if you want to keep full compatibility), once I realised it should be correctly inserted. In my case, the riser should be pushed towards the power button. 

[image]
[image]
[image]
[image]

klaudiawlodarskaa-ui, 2025-08-17:
Thank you so much!, especially to mr4lexndr, replacing the transistor and checking the connection in the riser helped. I am closing this issue!

closed 2025-08-17 by klaudiawlodarskaa-ui